emanuelhuber/RGPR #2: RD3 data not properly imported

jmerc13, 2017-09-05:
First, thanks for the package!

I am trying to import Mala data (.rd3), but have found that the resulting data (650 stacks) are just the same stack repeated over and over again. I've tried looking at the underlying code via: getMethod("readGPR", signature = c(fPath="character")) and RGPR:::readRD3. I get stuck debugging on lines 17 and 18: nTr <- .getHD(hRAD, "LAST TRACE") and nPt <- .getHD(hRAD, "SAMPLES"). I don't know enough about the details of R (or the S4 method) to know what the dot in front of the function getHD() means, or how to access its behavior. If you have suggestions I might be able to figure out how to do pull request and suggest a change to the code.

Cheers!

jmerc13, 2017-09-05:
I'm going to try and figure out how to do a pull request, but I think I found the problem. You specify your connection as the fNameRD3 object instead of the con object. Suggested edit:
con <- file(fNameRD3, "rb")
for (i in seq_len(nTr)) {
  dataRD3[, i] <- readBin(con, what = integer(), n = nPt, 
                          size = 2)
}

emanuelhuber, 2017-09-05:
Thank you jmerc13!
I updated the code with your suggestion and tested it.
Should correctly work!

closed 2017-09-05 by emanuelhuber