emanuelhuber/RGPR #3: Survey date of rd3 file populates current date, rather than actual survey date

jmerc13, 2017-09-05:
As the title suggests, I noticed the survey date (dd/mm/yyyy) of the imported metadata is today's date and not the date my data was actually collected. I think this may be related to the other issue I submitted.

Cheers!

emanuelhuber, 2017-09-05:
In the MALA header data file I have (".rad") there is no date (contrary to the Sensors&Software header data file). Therefore, I had decided to take the date of the file reading as the survey data, which is not quite correct.

Now I changed that. If there is no date in the header file, the following is done:

```r
x@date <- character(0)
```

When typing `x` (your current object of the `GPR` class) in the command console, the survey date is not shown if it is `character(0)`.

You can set the date by yourself using the new `svDate()` function:

```r
svDate(x)  # get the survey date (return NULL if there is no date
svDate(x) <- as.Date("2014-03-02")
svDate(x)
# or
svDate(x) <- as.Date(format(Sys.time(), "%Y-%m-%d"))
svDate(x)
```

One alternative (for later) would be to check if there is a coordinates file (".cor") and extract in this file the survey date... I add this alternative on my todo list.

jmerc13, 2017-09-05:
An alternative would to be extract the date and time information from the rd3 file, rather than the header:
`file.mtime("YourFile")`
Or if you want just the date:
`as.Date(as.character(file.mtime("YourFile")))`
This, of course, assumes that the date modified is the collection date.

emanuelhuber, 2017-09-05:
That's a possible alternative. It should be first checked, how the `file.time()` changes when the .rd3 file is copied at another place: does the date and time information stay the same? In other words: does the date and time information of the .rd3 file always reflect the survey date and time?

jmerc13, 2017-09-05:
I just completed a two week time lapse survey using Mala gear. All the dates and times relative to when I collected that data are intact. I copied and pasted those files to different places on my computer and it did not change anything. However, this information is likely to change if that data were downloaded, or transferred to another computer, as the original files would no longer be used.

emanuelhuber, 2017-09-09:
Thank you for testing. I prefer to not use the date of the data file for the survey date. I will explore another option: extract the date from the coordinate file ".cor" (this date would really correspond to the survey date). I will first check if all .cor files have the same structure/format...

jmerc13, 2017-09-11:
Yeah, that was the best thing I could think of, because going through all of my ancillary Mala files (.mrk, .em, .cor, .add, .rad) none of them have survey acquisition time and date. Maybe that is something to bring up with Mala? I noticed that not having this information generates an error when trying to use the GPS survey functions, because they require that bit of info. I was going to look into that code a bit later before adding it to an issue.

emanuelhuber, 2017-09-11:
Strange... do you know if there is an official documentation on the MALA file format?

jmerc13, 2017-09-11:
I'm asking some of the geophysicists that manage the equipment for the project I'm working on. Maybe they'll have some ideas. I may contact Mala directly to see if they have any insights.

jmerc13, 2017-09-12:
It sounds like Mala doesn't actually record that information in a metadata file; that metadata is not recorded in a specific file.

closed 2017-09-16 by emanuelhuber