emanuelhuber/RGPR #17: getArgs() in traceShift()

guifh, 2018-04-24:
Looks like there might be an issue with the function getArgs() in traceShift(). This error message only occurs when the traceShift() function is wrapped in another function. Everything is fine when I run the function alone. I couldn't figure out what is the exact problem...

Error in FUN(X[[i]], ...) : object 't0' not found
FUN(X[[i]], ...) 
FUN(X[[i]], ...) 
lapply(X = X, FUN = FUN, ...) 
sapply(arg[3:narg], eval) 
getArgs() 
traceShift(A2, ts = (-t0), method = "cubic", crop = FALSE)

guifh, 2018-04-24:
Same thing with time0Cor() actually.

emanuelhuber, 2018-04-25:
reproducible example:
```r
truc <- function(x, u = TRUE){
  getArgs()
}
truc2 <- function(x, b, v){
  truc(x = x,  b)
}

truc(x = 1:10, u = FALSE)
truc2(x = 1:10, b = FALSE, v = 1:10)
##  Error in FUN(X[[i]], ...) : object 'b' not found 
```

guifh, 2018-04-25:
Oui, exactement, ou si t0 est dans defini a l'interieur de la fonction:

```
truc2 <- function(x){
  tfb <- firstBreak(x)
  t0 <- firstBreakToTime0(tfb[1], x[,1])
  x <- traceShift(x, -t0, method = "cubic", crop = FALSE)
}

truc2(x)
## Error in FUN(X[[i]], ...) : object 't0' not found
```

emanuelhuber, 2018-04-25:
* I fixed the problem: now `getArgs()` returns something only if it is inside a function at level 1 (if the function calls itself another function that calls `getArgs()`, `getArgs()` will return nothing. That makes sense: we only want to know the first level processing (enough to make code reproducible).
* I also find a way to extract the function behind `FUN` (no need to use `getFunName()` anymore).
* I don't tested too much the changes... hopefully it will work well....

closed 2018-04-25 by emanuelhuber