emanuelhuber/RGPR #27: Problem installing RGPR

wilhem, 2020-07-17:
Hi guys,

I need to install the tool on my ubuntu 18.10 system. But I get the following error:

```
> devtools::install_github("emanuelhuber/RGPR")
Downloading GitHub repo emanuelhuber/RGPR@master
Skipping 1 packages not available: XML
Installing 41 packages: sp, rgdal, rgeos, signal, colorspace, Cairo, rgl, mmand, e1071, plot3D, adimpro, raster, fields, MBA, geojsonsf, sf, XML, checkmate, RConics, robfilter, httpuv, xtable, sourcetools, fastmap, shiny, miniUI, webshot, dotCall64, rapidjsonr, DEoptimR, manipulateWidget, misc3d, awsMethods, spam, maps, jsonify, sfheaders, classInt, DBI, units, robustbase
Installing packages into ‘/usr/local/lib/R/site-library’
(as ‘lib’ is unspecified)
Fehler: Failed to install 'RGPR' from GitHub:
  (converted from Warning) package ‘XML’ is not available (for R version 3.6.3)

```
What did I do wrong?

emanuelhuber, 2020-07-17:
Hello
Thank you for your question. May I suggest you to update R to the new version 4.0.
Check here if you need help (also for updating packages):
https://cran.r-project.org/ -> Download R for Linux -> Ubuntu -> check for the version corresponding to your ubuntu distribution.
You may need to install some libraries outside from R (like rgdal). If you encounter an error, simply google the solution. 
If you have any question let me know.
Let me know if your problem is solved.
Best,
Emanuel

emanuelhuber, 2020-07-17:
For your information: I just updated R with the 4.0 version and succeeded to install RGPR (Ubuntu 18.04).

wilhem, 2020-07-17:
Thank you very much.
I'm still having problems during the compilation of the packages with come with RGPR.
I'll keep you updated

emanuelhuber, 2020-07-19:
Thank you for the update. If you have any question or if you need a specific function, let me know (with RGPR you can already interpolate horizontal slices, I had not the time to write a tutorial about that).
Best,
Emanuel

closed 2020-07-19 by emanuelhuber