emanuelhuber/RGPR #28: White line in the plot

wilhem, 2020-07-19:
Hi everybody, 

I'm trying to understand what the white line on the [plot](https://ibb.co/rfsM27P) means.

[url=https://imgbb.com/][img]https://i.ibb.co/5xFvhMD/Bildschirmfoto.png[/img][/url]

I put a red arrow at about 3 ns on the left side of the plot. 
There is any explanation about the meaning of the white line? I couldn't find anything on github.

Another question: it seems that the speed of light is going to be calculated every time I plot something. How it is calculated?

Many thanks

emanuelhuber, 2020-07-19:
Hello
There shouldn't be a white line there. This could be a R issue. Try to resize the plot area and re-plot the data.

Concerning the speed of light, are you referring to the right y-axis? This axis is only here as an indication of the equivalent depth. It is based on the antenna separation distance and the speed of light (constant = 0.299 m/ns) as well as the EM wave propagation velocity in the media (that you can check/define with 'vel(x)').
Best,
Emanuel

emanuelhuber, 2020-07-20:
I close this issue. We can always re-open it if needed.
Best,
Emanuel

closed 2020-07-20 by emanuelhuber