emanuelhuber/RGPR #29: Function firstBreakToTime0() not clear

wilhem, 2020-07-20:
In the [website](http://emanuelhuber.github.io/RGPR/02_RGPR_tutorial_basic-GPR-data-processing/#time-zero-correction) about the time correction is reported that:

> Here we define
> t0=tfb−a/c0(1)
> 
> where a  is the distance between the transmitter and receiver and c0 is the wave velocity in the media between the transmitter and receiver (in our case, air)

but then the function firstBreakToTime0() is called:

> t0 <- firstBreakToTime0(tfb, x)

So now the question. Is _x_ passed as argument the distance _a_ between Tx and Rx? Or is _x_ the scan data read using the function _readGPR()_ ?

I tried both and I found out that:
putting the distance between Tx and Rx (in my case 0.08 m), I get:

> Fehler in firstBreakToTime0(tfb, 0.08) : 
>   versuche einen Slot "antsep" von einem Objekt der einfachen Klasse ("numeric") ohne Slots anzufordern

putting the scan data read by _readGPR()_ I get an useful t0 but to me it doesn't make the explanation above any sense

Thanks!

emanuelhuber, 2020-07-20:
Thank you very much for your feedback, it helps me to improve RGPR.

You have to pass the GPR object (output from 'readGPR()') to the function 'firstBreakToTime0'. Why? Because the GPR object contains the information about the antenna separation distance (the 'a' in the equation). Note that you can use 'estimateTime0()' instead of 'firstbreak()' + 'firstBreakToTime0()'.
Is it clearer for you?

wilhem, 2020-07-21:
Hi Emanuel,

you closed the issue on github too fast 😉
Can you please tell me which field is the antenna separation distance (the
*a*) in the GPR object? Here a GPR object from my dataset:

> str(x1)
> Formal class 'GPR' [package "RGPR"] with 29 slots
>   ..@ version     : chr "0.2"
>   ..@ data        : num [1:1022, 1:490] -0.00153 -0.00153 -0.00153
> -0.00153 -0.00153 ...
>   ..@ traces      : int [1:490] 1 2 3 4 5 6 7 8 9 10 ...
>   ..@ depth       : num [1:1022] 0 0.0196 0.0391 0.0587 0.0782 ...
>   ..@ pos         : num [1:490] 0 0.005 0.01 0.015 0.02 0.025 0.03 0.035
> 0.04 0.045 ...
>   ..@ time0       : num [1:490] 0 0 0 0 0 0 0 0 0 0 ...
>   ..@ time        : num [1:490] NaN Inf Inf Inf Inf ...
>   ..@ fid         : chr [1:490] "A001" "A002" "A003" "A004" ...
>   ..@ ann         : chr(0)
>   ..@ coord       : logi[0 , 0 ]
>   ..@ rec         : logi[0 , 0 ]
>   ..@ trans       : logi[0 , 0 ]
>   ..@ coordref    : num(0)
>   ..@ freq        : num 500
>   ..@ dz          : num 0.0196
>   ..@ dx          : num 0.005
>   ..@ antsep      : num 0
>   ..@ name        : chr "metal_pipe_merged"
>   ..@ description : chr ""
>   ..@ filepath    : chr "rawGPR/metal_pipe_merged.dzt"
>   ..@ depthunit   : chr "ns"
>   ..@ posunit     : chr "m"
>   ..@ surveymode  : chr "reflection"
>   ..@ date        : chr "2020-07-21"
>   ..@ crs         : chr(0)
>   ..@ proc        : chr(0)
>   ..@ vel         :List of 1
>   .. ..$ : num 0.106
>   ..@ delineations: list()
>   ..@ hd          :List of 24
>   .. ..$ TAG       : int 255
>   .. ..$ OFFSETDATA: num 1024
>   .. ..$ NSAMP     : int 1024
>   .. ..$ BITS      : int 16
>   .. ..$ ZERO      : int -32768
>   .. ..$ SPS       : num 0
>   .. ..$ SPM       : num 200
>   .. ..$ MPM       : num 0
>   .. ..$ POSITION  : num 0
>   .. ..$ RANGE     : num 20
>   .. ..$ NPASS     : int 0
>   .. ..$ DATE      : chr "1980-00-00"
>   .. ..$ TIME      : chr "00:00:00"
>   .. ..$ OFFSETTEXT: int 0
>   .. ..$ NTEXT     : int 0
>   .. ..$ PROC      : int 0
>   .. ..$ NPROC     : int 0
>   .. ..$ NCHAN     : int 1
>   .. ..$ EPSR      : num 8
>   .. ..$ TOP       : num 0
>   .. ..$ DEPTH     : num 1.06
>   .. ..$ ANT       : chr "gprMax 500MHz "
>   .. ..$ VSBYTE    : int 0
>   .. ..$ MRKS      : num [1:490] -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 ...
>

Thanks!

*Davide Picchi*

Mobil: +49 176 86308958
paveway@gmail.com

emanuelhuber, 2020-07-21:
The antenna separation is stored in the slot `@antsep`. You can access this value with:

`antsep(x)`

and you can set a new value (for example 2) with:

`antsep(x) <- 2`

Where `x` is a GPR object. 

When you read a GPR data, RGPR try to extract the antenna separation distance from the data file.

wilhem, 2020-07-22:
Dear Emanuel,

I wonder how fast you reply to questions from users. That's really
respectable. So many thanks for your help and your explanations.
Anyway I have some real data files from a GSSI Structure Scan XT device
(here attached). To me the antenna separation should be about 80 mm in that
device.
But importing datas I get 0:

> str(x1)
> Formal class 'GPR' [package "RGPR"] with 29 slots
>   ..@ version     : chr "0.2"
>   ..@ data        : num [1:510, 1:903] 0.00041 0.000358 0.00031 0.000349
> 0.000295 ...
>   ..@ traces      : int [1:903] 1 2 3 4 5 6 7 8 9 10 ...
>   ..@ depth       : num [1:510] 0 0.00803 0.01607 0.0241 0.03214 ...
>   ..@ pos         : num [1:903] 0 0.00125 0.0025 0.00375 0.005 ...
>   ..@ time0       : num [1:903] 0 0 0 0 0 0 0 0 0 0 ...
>   ..@ time        : num [1:903] 1.43e+09 1.43e+09 1.43e+09 1.43e+09
> 1.43e+09 ...
>   ..@ fid         : chr [1:903] "" "" "" "" ...
>   ..@ ann         : chr(0)
>   ..@ coord       : logi[0 , 0 ]
>   ..@ rec         : logi[0 , 0 ]
>   ..@ trans       : logi[0 , 0 ]
>   ..@ coordref    : num(0)
>   ..@ freq        : num 0
>   ..@ dz          : num 0.00803
>   ..@ dx          : num 0.00125
>   ..@ antsep      : num 0
>   ..@ name        : chr "FILE__008"
>   ..@ description : chr ""
>   ..@ filepath    : chr "rawGPR/FILE__008.DZT"
>   ..@ depthunit   : chr "ns"
>   ..@ posunit     : chr "cm"
>   ..@ surveymode  : chr "reflection"
>   ..@ date        : chr "2015-03-05"
>   ..@ crs         : chr(0)
>   ..@ proc        : chr(0)
>   ..@ vel         :List of 1
>   .. ..$ : num 0.12
>   ..@ delineations: list()
>   ..@ hd          :List of 24
>   .. ..$ TAG       : int 2047
>   .. ..$ OFFSETDATA: num 131072
>   .. ..$ NSAMP     : int 512
>   .. ..$ BITS      : int 32
>   .. ..$ ZERO      : int 3
>   .. ..$ SPS       : num 250
>   .. ..$ SPM       : num 800
>   .. ..$ MPM       : num 5
>   .. ..$ POSITION  : num -0.411
>   .. ..$ RANGE     : num 4.11
>   .. ..$ NPASS     : int 0
>   .. ..$ DATE      : chr "2015-03-05"
>   .. ..$ TIME      : chr "27:01:00"
>   .. ..$ OFFSETTEXT: int 512
>   .. ..$ NTEXT     : int 0
>   .. ..$ PROC      : int 128
>   .. ..$ NPROC     : int 13
>   .. ..$ NCHAN     : int 1
>   .. ..$ EPSR      : num 6
>   .. ..$ TOP       : num 0.0246
>   .. ..$ DEPTH     : num 0.246
>   .. ..$ ANT       : chr "none"
>   .. ..$ VSBYTE    : int 28672
>   .. ..$ MRKS      : int [1:903] 0 0 0 0 0 0 0 0 0 0 ...
>

could be that not every device is going to write that parameter into the
file?
Which measure unit has that parameter? Meter [m] or millimeter [mm]. In
other words: should I put there 0.08 or 80 if the antenna separation 80 mm
is?

Thank you very much.
Davide

*Davide Picchi*

Mobil: +49 176 86308958
paveway@gmail.com

wilhem, 2020-07-22:
Sorry again.
I just wanted to add, that trying to extimate t0 from the same dataset in
my previous email, leads to the following error:

> x2 <- estimateTime0(x1, method = "coppens", thr = 0.05, w = 5, FUN = mean)
> Fehler in checkArgStop(msg) : Check arg
> arg 'w': must be a numeric value > 0 and < 0
>

Even putting w = 0, the error is the same

*Davide Picchi*

Mobil: +49 176 86308958
paveway@gmail.com

emanuelhuber, 2020-07-22:
Hello Wilhem

I want to be reactive to help users and keep them using RGPR (with the objective to improve RGPR).

- **antenna separation distance information in the binary file:** each manufacturer has its own binary file format. Some are very straightforward others are messy. I cannot tell you what is the problem with your file but if you send it to me, I can check that and maybe improve the function (I never share data from users with third parties without explicit authorisation, Swiss confidentiality).
- **unit of antenna separation distance:** you should use the same unit as for the trace coordinates (I assume you should use meters)
- **error estimating time-zero:** the algorithm only examines the part of the signal before the largest amplitude (because of the assumption that the first wave break should occur before the largest amplitude). In you case, your signal seems to start with the largest amplitude at 0 ns and therefore the function stucks (nothing to find between 0 ns ans 0 ns). I should add an option in the code to ignore the largest amplitude at the signal beginning. In the mean time, you can discard the part of the signal with the largest amplitude. If the largest amplitude is only stored in the first row of your data, you can do the following (remove first row when estimating time0).
    ```
    x2 <- estimateTime0(x1[-1, ], ...
    ```
If this does not work, you can send me the data for a check. 

Best,
Emanuel

emanuelhuber, 2020-07-23:
I fixed the bug with `estimateTime0()`. It should now work. If not, I will re-open this issue.

closed 2020-07-23 by emanuelhuber