emanuelhuber/RGPR #30: Object fdo not found

wilhem, 2020-07-22:
Trying to estimate the fresnel zone I get the following error:

> spec2D(x4):fdo
Fehler: Objekt 'fdo' nicht gefunden
> spec(x4):fdo
Fehler: Objekt 'fdo' nicht gefunden

The issue arises even using the method _spec2D()_ instead of _spec()_.
Any idea, how to fix it?
Thanks

emanuelhuber, 2020-07-22:
Very strange, the object `fdo` appears only in the function `migrate`... can you show me the code you are using? Did you install the most recent version of RGPR?

wilhem, 2020-07-22:
Ok.
I have a better analysis of the problem now.

My RGPR version is:

```
> version
               _                           
platform       x86_64-pc-linux-gnu         
arch           x86_64                      
os             linux-gnu                   
system         x86_64, linux-gnu           
status                                     
major          4                           
minor          0.2                         
year           2020                        
month          06                          
day            22                          
svn rev        78730                       
language       R                           
version.string R version 4.0.2 (2020-06-22)
nickname       Taking Off Again  
```

The data I'm using is the data from 2014_04_25_frenke.zip
However, here a better explanation: going through the tutorial on this page https://emanuelhuber.github.io/RGPR/05_RGPR_tutorial_GPR-data-migration/#pre-processing
you process data to get _A8_:

`A8 <- upsample(A7, n = c(3,1))`

but 3 lines below you need the 'fdo' parameter for running the kirchhoff migration algorithm:

`spec(A9): fdo = 80`  (1)

Now is the problem clear: going through the tutorial you get _A8_. But for the 'fdo' you need another process step: `A9`
And this is unfortunately not clear, how to get that, since you get A9 running the Kirchhoff migration:

`A9 <- migration(A8, type="kirchhoff", max_depth = 20,  dz = 0.01, fdo = 80)`   (2)

So it is like a chicken-egg problem.
You need 'fdo' to run the migration algorithm (1), but the migration algorithm needs 'fdo' (2).

I hope I could help

emanuelhuber, 2020-07-22:
Alright, I understand the problem (a communication problem). In the tutorial `spec(A9): fdo = 80` means: "if you look at the spectrum of `A9` using the function `spec()`, you will see that the return frequency is about `80` MHz". Therefore, you set `fdo = 80` in the function `migrate()`.
I will re-write the tutorial to make it clearer.

emanuelhuber, 2020-07-28:
The tutorial is updated.

closed 2020-07-28 by emanuelhuber