emanuelhuber/RGPR #32: Reading Sensors & Software GPZ files

pwernette, 2020-07-23:
So pleased to see an open-source alternative to GPR data processing. I was just curious if you have thought about adding the ability to read Sensors and Software files with a .GPZ extension? I'm familiar with their .hd and .DT1 files, but wasn't sure if it would be possible to read the newer GPZ files.

agianno, 2020-07-24:
@pwernette As far as I know the GPZ file must be opened by EKKO Project as it is encrypted. It seems to be an encrypted zip file. Once opened you can get the .DT1 and .HD files

emanuelhuber, 2020-07-27:
Thank you @agianno for your comment. Currently, RGPR does not support .gpz files and I don't think that RGPR will support such kind of files in a near future. It seems really challenging to read such files. According to Sensors and Software:

> EKKO_Project works with project .gpz files automatically exported from current GPR systems. An embedded database structure allows all GPR data and ancillary files (GPS, topography, photos, notes, etc.) to be housed in a single .gpz file for a specific project. A project file could contain a single GPR line or several hundred lines from a complex survey. Legacy .dt1 and .hd files can also be imported. Any available project information can easily be added to the .gpz file, providing a single point for all your survey information.

closed 2020-07-27 by emanuelhuber