emanuelhuber/RGPR #33: Possible misunderstanding in the tutorial

wilhem, 2020-07-28:
Hi Emanuel,

according to the output of the function `str()`, you should get the parameter `dz`, which is the sampling rate of the radar

```
> str(A)
Formal class 'GPR' [package "RGPR"] with 29 slots
  
  ..@ dz          : num 0.00893   # Sampling rate
```

But in the migration [section](https://emanuelhuber.github.io/RGPR/05_RGPR_tutorial_GPR-data-migration/#first-wave-break-estimation-and-set-time-zero) `dz` is the vertical resolution and its unit is "m"

> Topographic Kirchhoff migration.
> Vertical resolution of the migrated data, we set dz = 0.01m.

Is it just a coincidence and `dz` means two different things? Or they are the same thing and dz is the sampling rate even in the second case?

emanuelhuber, 2020-07-28:
Hello wilhem

It is intentionally that `dz` can be in time unit or in depth unit. That is maybe not very elegant. The challenge I faced was that a GPR trace can be a signal amplitude as a function of time (the recorded signal) or a signal as a function of depth for migrated GPR data. I did not want to handle two different slots for a GPR object (one for the sampling rate in time unit (e.g., `dt`) and another one for the sampling rate in depth unit `dz`). That's the explanation.
Note that the standard user should not notice that because the slot `dz` has no setter/getter. In the develop version of RGPR, the slot `dz` as well as `depthunit()` will disappear and unit conversion will be supported.

In your data, `dz` is most likely in time unit (check the output of `depthunit(A)`). In the migration function, `dz` correspond to the vertical resolution in depth unit (e.g., meters) you want to obtain (because the output of `migrate()` in a signal as a function of depth).

Does it answer your questions?

closed 2020-07-29 by emanuelhuber

wilhem, 2020-08-02:
Thank for your reply. That's clear the question.

But I found another possible misunderstanding regarding the function `dewow()`
In the wiki it is reported that:

> **Dewow**

> Remove the low-frequency components (the so-called “wow”) of the GPR record using:
> 
>     a running median filter (type = "runnmed")
>     a running mean filter (type = "runmean")
>     a Gaussian filter (type = "Gaussian")
> 
> For the two first cases, the argument w is the length of the filter **in time units**.

But looking at the help

`?dewow`

I get:

> w	[numeric(1)] If type = runmed, MAD or runmean, window length of the filter **in trace unit**; If type = Gaussian, standard deviation in trace unit. If w = NULL, w is estimated as five times the wavelength corresponding to the maximum frequency of x (estimated with spec)

Now it is not clear to me what _w_ should be. 

Let's say I have in my function `w = 5` (time unit [ns]).
Does it mean that the window length of the filter goes from 0 _ns_ to 5 _ns_ all over the traces?
Or does it mean, that the window length of the filter for every _n_ trace [n-5, n-4, n-3, ...0, n+1, n+2,.., n+5] goes from 0 _ns_ till the end of the vertical time windows?

Regards