emanuelhuber/RGPR #34: Taking the firstBreak() function inspect

wilhem, 2020-07-28:
Dear Emanuel,

I was taking a look to the function:`firstBreak()`
and found something "strange".

Let's say I have the following signal (1D plot) from my device:

[image]

As you can see the signal is going to cross the +/- 0.005 mV line (just image them ;) ) in the time window from 0 to 1 ns.

Running the function firstBreak() as following:

`firstBreak(x1, w = 1.5, method = "threshold", thr = 0.005 )`

I would expect that, since `thr = 0.005`, the output of this function should be something like (for the trace on the picture): 
t1 ~= 0.75 ns (for +0.005 mV)
t2 ~= 1.25 ns (for -0.005 mV)

But I get only a vector with all Zeros (for all traces)....

Rising the value of thr up to 0.5:

`firstBreak(x1, w = 1.5, method = "threshold", thr = 0.5 )`

I get a vector like [0.423, 0.441, 0.448 ..] anyway... **_not_** Zero.

That's really strange to me, since, as the 1D plot shows, the signal is never crossing the +/-0.5mV lines....

Why rising the threshold I get useful values from the function?

Many many thanks

guifh, 2020-07-28:
Dear Wihlem,

Please, note that, as specified in the help of the firstBreak function, the threshold should be given in % (that is % value of your maximum amplitude). We recommend a value between 10% and 20%. If you ask the function to pick time zero on a 0.005% threshold you will indeed get value equal or very close to 0.

guifh, 2020-07-28:
Dear Wilhem,

Please, note that, as specified in the help of the firstBreak function, the threshold is given in % (that is % of the maximum amplitude). We recommend a value between 0.1 (10%) and 0.2 (20%). By specifying 0.005, you ask the function to pick time zero on a 0.5% threshold, which will indeed return values equal or close to 0. With 0.5, you ask to pick time zero on a 50% threshold, hence the values close to ~0.4 returned.

emanuelhuber, 2020-07-28:
Precision: with the threshold method, the sample before the first sample that is larger than the threshold times the maximum **absolute** amplitude is picked. Then the time of the first wave break is linearly interpolated between these two samples.
I updated the documentation of `firstBreak()` precising that 0 correspond to 0% and 1 to 100%, and I also added explanations about the three methods available.

closed 2020-07-28 by emanuelhuber