emanuelhuber/RGPR #36: Possible bug in the function spec()

wilhem, 2020-08-03:
Dear guys,

I recognize something that to me doesn't make any sense. Perhaps the function `spec()` needs to be checked again.

The problem is the following:
I load a dataset using:

`A1 <- readGPR( dsn = filename, verbose = TRUE )`

and then I apply the DCshift removal, dewow, and time zero correction. So I'm now at this step:

`A4 <- time0Cor( A3, method = "linear" )`

doing:

`spec(A4) `

I get the following spectrum:

[image]

As you can see the spectrum starts from _0 Hz_, which is correct. The peak frequency is about 1500-2000 MHz, which is also correct.

Now let's do another step:

`A5 <- timeCorOffset(A4)`

which shifts all the traces to the zero.

Doing again the same command as before:

`spec(A5)`

I get the following spectrum:

[image]

Please note that in the second plot the frequency was shifted to zero! Now is the peak around 0 Hz, which doesn't make any sense. A time zero correction should move the time axis to 0, but the frequency should stay the same and not be changed through the time zero correction.
To me there should be a difference plotting:

```
plot(A4)
plot(A5)
```

but not plotting:
```

spec(A4)
spec(A5)
```

Am I wrong? Or the function` spec()` should be rechecked?

emanuelhuber, 2020-08-03:
Hello Wilhem 

Please note that `timeCorOffset()` is not the same as `time0Cor()` that shifts the traces to time-zero. 

`timeCorOffset()` corrects each trace for the offset between transmitter and receiver antennae: it converts the trace time of the data acquired with a bistatic antenna system into trace time data virtually acquiered with a monostatic system. In fact, a normal move-out (NMO) correction is applied, a non-linear operation. That means that `timeCorOffset()` will modify the frequency spectrum.

I will test what you did and check if something is weird in the `spec()` function (I don't think so, because this function is based on the internal R-function `fft()`) or if I can find an explanation for the frequency spectrum of the output of `timeCorOffset()`.

wilhem, 2020-08-04:
Thanks for your reply but it doesn't make sense to me to have a peak frequency at about 0 Hz, even if the system collapse to an equivalent monostatic antenna. Analysing the spectrum of the reflected signal I shouldn't have any frequency shift, since the reflected frequency is dependent on the electro-magnetic properties of the material, in which the radiowaves travel.

I'll check again the documentation in R

wilhem, 2020-08-04:
I wanted just to give an update:
typing:

`?timeCorOffset`

it opens the help page of

> firstBreakToTime0 {RGPR}	R Documentation
> Convert first wave break to time-zero
> Description
> Account for the delay time between time of wave emission and time of first wave break recording due to the antenna separation (offset)....

So I cannot get any help online explanation about the function `timeCorOffset()`

emanuelhuber, 2020-08-12:
Hello Wilhem

The frequency spectrum of GPR data is not necessarily easy to interpret (except for the noise on the right side of the spectrum) because a GPR signal is a mixture of many cosinus functions with significant amplitudes. 

I wrote a short code to 

1. visualise the effect of processing on the power spectrum (I did not get similar results but that is normal because I do not use the same data.
2. decompose a GPR trace into a series of cosinus functions based on the Fourier transform (and on the `spec()` function). The reconstruction of the signal shows that the `spec()` functions is correctly implemented. If you need some mathematical background on the discrete Fourier transform and its back-transform, you can check this excellent textbook freely available as PDF: http://www.dspguide.com/ch8/4.htm

You can try to apply this code to your data and to understand the relationship power spectrum <-> signal.

```
devtools::install_github("emanuelhuber/RGPR")
library(RGPR)

x <- frenkeLine00

# Plot power spectrum of trace i (by default i = 10)
plotSpec <- function(x, i = 10, type = "l", ...){
  x_spec <- spec(x[,i], plot = FALSE)
  plot(x_spec$freq, x_spec$pow, type = type, ...) 
}

par(mfrow = c(2, 1))
plot(x[,10])
plotSpec(x, xlab = "freq", ylab = "power", main = "Power spectrum")

#------------------------------------------------------------------------------#
# Check processing effect on the power spectrum
par(mfrow = c(5, 1))
plotSpec(x, xlab = "freq", ylab = "power", 
         main = "raw data")

x1 <- dcshift(x)
plotSpec(x1, xlab = "freq", ylab = "power",
         main = "after DC-shift")

x2 <- estimateTime0(x1, method = "coppens", w = 5 )
x2 <- time0Cor(x2)
plotSpec(x2, xlab = "freq", ylab = "power", 
         main = "after time-zero correction")

x3 <- dewow(x2, w = 100)
plotSpec(x3, xlab = "freq", ylab = "power", 
         main = "after dewow")

x4 <- timeCorOffset(x3)
plotSpec(x4, xlab = "freq", ylab = "power", 
         main = "after timeCorOffset")

#------------------------------------------------------------------------------#
# function to decompose a trace of a GPR object into cosinus functions
# based on the Fourier transform
# for the mathematical background, check: http://www.dspguide.com/ch8/4.htm
# Note: if ncol(x) > 1, only the first trace will be used
cosWaveTrans <- function(x){
  cosw <- function(freq, pha, A, n, tt){
    A * cos(2 * pi * freq*10^6 * tt*10^-9 + pha )/n
  }
    
  x_spec <- spec(x, plot = FALSE, unwrapPhase = FALSE)
  tt <- depth(x)
  vn <- rep(length(x_spec$freq), length(x_spec$freq))
  vn[1] <- nrow(x)
  vn[length(vn)] <- vn[1]
  cos_wave <- mapply(cosw, x_spec$freq, x_spec$pha[, 1], x_spec$pow[, 1], vn, MoreArgs = list(tt = tt))
  return(list(x = tt, y = cos_wave, freq = x_spec$freq))
}

# decompose the 10th trace into a series of cosinus
cos_wave <- cosWaveTrans(x[,10])

par(mfrow = c(1,1))
plot(x[,10])
lines(cos_wave$x, rowSums(cos_wave$y), type = "l", col = "red")

#------------------------------------------------------------------------------#
# plot some single harmonics
v <- c(1, 2, 5, 50, 10)
par(mfrow = c(length(v), 1))
for(i in seq_along(v)){
plot(cos_wave$x, cos_wave$y[, v[i]], type = "l", 
     ylab = "Amplitude", xlab = "Time",
     main = paste0(cos_wave$freq[v[i]], " MHz"))
}

#------------------------------------------------------------------------------#
# plot the sums of harmonics from the first one
v <- c(50, 100, 150, 200, 250)
par(mfrow = c(length(v), 1))
for(i in seq_along(v)){
  plot(cos_wave$x, rowSums(cos_wave$y[,1:v[i]]), type = "l", 
       ylab = "Amplitude", xlab = "Time",
       main = paste0("0 - ", cos_wave$freq[v[i]], " MHz"))
       # ylim = range(x) * 1.5)
}

#------------------------------------------------------------------------------#
# plot the sums of a range of harmonics
v <- c(1, 50, 100, 150, 200, 250)
par(mfrow = c(length(v) -1, 1))
for(i in seq_along(v[-1])){
  plot(cos_wave$x, rowSums(cos_wave$y[,v[i]:v[i+1]]), type = "l", 
       ylab = "Amplitude", xlab = "Time",
       main = paste0(cos_wave$freq[v[i]]," - ", cos_wave$freq[v[i+1]], " MHz"))
       # ylim = range(x))
}
```

emanuelhuber, 2020-08-12:
And note that the cosine wave at zero frequency is constant. That means that the real part of the discrete Fourier transform at frequency zero holds the average value of all the points in the time domain signal.

In chapter 9, page 172 of the book "The Scientist and Engineer's Guide to Digital Signal Processing" (http://www.dspguide.com/CH9.PDF) Steven W. Smith briefly presents what he called the 1/f noise, also called pink noise (see wikipedia) whose origin is a mystery: The spectrum amplitudes increase with decreasing frequency. No idea the processing step you applied (`timeCorOfffset()`) created this 1/f noise...

Could you please show a common trace of `A4` and `A5`: that maybe holds the explanation for that noise.

By the way, the help page of the function `timeCorOffset()` is now working.

emanuelhuber, 2020-08-30:
I close this issue. If you have any comments or questions, we can always open it again.
Best,
Emanuel

closed 2020-08-30 by emanuelhuber

wilhem, 2020-09-21:
Thank you very much for your explanation.
I'm sorry to answer so late to your post above, but I couldn't concentrate on this topic for a while.

Despite your explanation and your help there is still something (to me) misleading. I try to put in a very simple way. As a base let's take the tutorial on this [site](https://emanuelhuber.github.io/RGPR/07_RGPR_tutorial_GPR-data-migration/), so we avoid to speak about different datas.
Going through the tutorial and doing
`spec(A8) `
we get an amplitude peak around frequency 80 MHz.

That's ok.

But for some reasons, using another radar data coming from GSSI and making the same steps I get that the peak is at 0 MHz, which of course cannot be possible.

So the question is:
the spec(x) command should be done after sampling the data (after upsample)??
Or can I run that command earlier? For istance after time corretion?

In the second case I don't get a max amplitude at the 0 Hz frequency. For this reason it makes sense to me to take that frequency.
Or am I wrong?

Thanks a lot for your time and patience

emanuelhuber, 2020-09-21:
The `spec()` function can be applied at any time (it is just used to have a look at the spectrum). It is not a processing function.

It is hard to diagnostic your data without knowing how they look like. Did you adapt the argument values of the processing function to your data? If you have the largest amplitude toward the smallest frequency that could mean, that you have some low frequency "noise". I would suggest you to remove these low-frequency signal with the `dewow()` function for example.

If you want, you can send me your data such that I can check what is happening and if your processing can be improved.

Normally, the dominant return frequency is slightly smaller that the antenna frequency. **Never set the dominant return frequency equal to 0 MHz in `migrate()`** else the Fresnel zone will be equal to zero and you will not migrate the data. The Fresnel zone is computed accordingly to this paper: Pérez-Gracia et al. (2008) Horizontal resolution in a non-destructive shallow GPR survey: An experimental evaluation. NDT & E International, 41(8): 611–620. doi:10.1016/j.ndteint.2008.06.002 

**Note:**
I compared the frequency spectrum returned by `spec()` with the spectral density returned by `spec.gram()` (base function used in `spectrum()`), they are nearly identical (see code below). Note that for the comparison the amplitudes returned by `spec()` must be squared and scaled but I still don't understand how). I would say that RGPR correctly computes the spectrum.

```
u <- A8[,10]
u_spec <- spec(u, plot = FALSE, unwrapPhase = FALSE)
u_spec.pgram <- spec.pgram(x = as.vector(u), fast = FALSE, demean = FALSE, detrend = FALSE,
                     plot = FALSE)

par(mfrow = c(2, 1))
plot(u_spec.pgram$spec, type = "l", pch = 20)

plot(u_spec$pow[-1, ]^2/length(u_spec$pow[-1, ])/2, type = "l", pch = 20)

par(mfrow = c(1, 1))
plot(u_spec.pgram$spec, u_spec$pow[-1, ]^2, asp = 1)
```

**Note:**
The amplitude at exactly 0 MHz corresponds to the signal average/mean (see the code below). I saw that this value is generally not displayed in power spectrum (maybe because it is confusing).

```
u <- A8[,10]

par(mfrow = c(1, 2))
plot(u)
lines(u - mean(u), col = "red") #r emove the mean

u_spec <- spec(u, plot = FALSE, unwrapPhase = FALSE)
plot(u_spec$pow, type = "o", pch = 20)

# remove the mean
udetrend_spec <- spec(u - mean(u@data), plot = FALSE, unwrapPhase = FALSE)
lines(udetrend_spec$pow, type = "o", pch = 20, col = "red")
```