emanuelhuber/RGPR #42: Header definition problem in readIDS.r

NeilParry, 2021-02-23:
Thanks for the quick attension to the interger typo bug I noted.
 
Seems to be another issue in the actual routine however as I am now getting a header error.
 
> x <- readGPR(dsn = "LID10001.dt", verbose = TRUE)
'R' not yet implemented!
Error in DDhd1[i] <- readBin(dsn, what = integer(), n = 1, size = 2) :
  replacement has length zero.
 
I have attached the IDS file I am reading in as my guess is there is a formatting difference between the IDS file and what the routine is expecting.
 
Regards
 
Neil

[LID10001.dt.zip](https://github.com/emanuelhuber/RGPR/files/6031428/LID10001.dt.zip)

NeilParry, 2021-02-23:
The full file likely didn't upload due to the 4Mbyte file limit. I can provide a dropbox link if I have an email address

emanuelhuber, 2021-02-23:
Hello Neil
Thank you, I got the file. I ,
could fix the problem. It should now work. Let me know if there is anything.
Best,

Emanuel

closed 2021-02-23 by emanuelhuber