emanuelhuber/RGPR #44: I can't import data

aquiles300, 2021-03-18:
I am trying to import MALA GPR data (*.rd3 format) through

>library(RGPR)
>x <- readGPR(dsn = "erickprueba250mhz10.rd3")

but I got

Error in matrix(NA, nrow = npt, ncol = ntr) : 
  invalid 'ncol' value (too large or NA)
In addition: Warning message:
In strsplit(headRAD[i], ":") : input string 1 is invalid in this locale

I dont  know what is wrong with that. I am sending the files, if you can help me with that.

[erickprueba250mhz10.zip](https://github.com/emanuelhuber/RGPR/files/6167571/erickprueba250mhz10.zip)

Marcel

emanuelhuber, 2021-03-18:
Thank you very much for your message, Marcel. It is hard to say what is the problem. I will have a closer look at your data and I hope to be soon able to solve this issue. Thank you for your patience.
Best,
Emanuel

emanuelhuber, 2021-03-18:
In the mean time, could you please show me the output of your session info:

```
utils::sessionInfo()
```

Thank you.

aquiles300, 2021-03-18:
Ok, thank you, emanuel. The output is

R version 4.0.4 (2021-02-15)
Platform: x86_64-redhat-linux-gnu (64-bit)
Running under: Fedora 32 (Workstation Edition)

Matrix products: default
BLAS/LAPACK: /usr/lib64/libopenblas-r0.3.12.so

locale:
 [1] LC_CTYPE=en_US.UTF-8       LC_NUMERIC=C              
 [3] LC_TIME=en_US.UTF-8        LC_COLLATE=en_US.UTF-8    
 [5] LC_MONETARY=en_US.UTF-8    LC_MESSAGES=en_US.UTF-8   
 [7] LC_PAPER=en_US.UTF-8       LC_NAME=C                 
 [9] LC_ADDRESS=C               LC_TELEPHONE=C            
[11] LC_MEASUREMENT=en_US.UTF-8 LC_IDENTIFICATION=C       

attached base packages:
[1] stats     graphics  grDevices utils     datasets  methods   base     

other attached packages:
[1] RGPR_0.0.7

loaded via a namespace (and not attached):
[1] compiler_4.0.4  magrittr_2.0.1  sp_1.4-5        grid_4.0.4     
[5] lattice_0.20-41

emanuelhuber, 2021-03-19:
Thank you Marcel
I could identify the problem: the encoding of the text file "erickprueba250mhz10.rad". It would be too complex to fix this issue in RGPR because it could have side effects that I am not yet aware of them. 
My suggestion: simply open the file "erickprueba250mhz10.rad" in a text editor and save it with encoding UTF-8 (without BOM). In windows, I recommend you to use Notepad++ (ask google how to change the encoding, it is straightforward).
Let me know if it solve your problem.

Let me know if you have any question, suggestion or feedback.

Best,

Emanuel

closed 2021-03-19 by emanuelhuber

aquiles300, 2021-03-19:
Hello

It works perfectly. Thank you

Marcel