emanuelhuber/RGPR #46: Error While Reading .DZT File

h2maa, 2021-06-07:
Dear Mr. Emanuel Huber,

While reading the .dzt file using your RGPR codes, I received the below error:

**Error in UseMethod("xmlValue") : 
  no applicable method for 'xmlValue' applied to an object of class "list"**

I would appreciate if you could help to debug this.

Thank you very much!

emanuelhuber, 2021-06-08:
Hello h2maa

Thank you very much for raising this issue that is most likely related to your .dzx file (if you remove this file from the directory this error message should disappear). I would like to fix this issue. Could you please send me your .dzx file (email: emanuel.huber@pm.me) such that I can reproduce the error and understand the problem?
Thank you,
Best,

Emanuel

h2maa, 2021-06-08:
Dear Mr. Emanuel Huber,

It works when I removed the .dzx file. Thank you so much!

I will send you this file to your email shortly.

emanuelhuber, 2021-06-08:
Hello

I fixed this issue (at least, I can read without any error message your .dzx file). Note that you need to re-install RGPR:

```r
if(!require("devtools")) install.packages("devtools")
devtools::install_github("emanuelhuber/RGPR")
library(RGPR)       # load RGPR in the current R session
```

Let me know if it now works. Best,
Emanuel

h2maa, 2021-06-10:
Dear Mr. Emanuel Huber,

Many thanks for your effort to fix the program. I reinstalled the package following your instruction but got the same error. I could send you both .dzt and .dzx files for your references.

Thank you.

h2maa

emanuelhuber, 2021-06-10:
Hello h2maa

Thank you very much for sending me the files. I tested and I could successfully read your data with RGPR. Please make sure that your re-started R before trying again.

The code below is what I used, also to export the GPR data to a .txt file (note that you need to adapt the file paths to your situation):
```r
if(!require("devtools")) install.packages("devtools")
devtools::install_github("emanuelhuber/RGPR")
library(RGPR)       # load RGPR in the current R session

dsn <- "/mnt/data/RGPR/CODE/DEVELOPMENT/FILE_FORMAT/dzt/h2maa_check_dzx/TRIAL__001.DZT"
x <- readGPR(dsn)
plot(x)

# export GPR data as .txt file (matrix-like format with 
# column names equal to trace position and row name equal to depth/time)
writeGPR(x, fPath = "/mnt/data/RGPR/CODE/DEVELOPMENT/FILE_FORMAT/dzt/h2maa_check_dzx/TRIAL_001.txt", type = "ASCII")

# export GPR data as .txt file (3-column ".txt" file. The columns
# correspond to 1) trace position, 2) time/depth, and 3) amplitude)
writeGPR(x, fPath = "/mnt/data/RGPR/CODE/DEVELOPMENT/FILE_FORMAT/dzt/h2maa_check_dzx/TRIAL_001.txt", type = "xta")
```

Let me know if it works and don't hesitate to contact me if you need any help. Any feedback is of course warmly welcomed.

Best,

Emanuel

emanuelhuber, 2021-06-11:
Very strange. Here is what I recommend you: close all R applications, re-start R, and run ony this 3 lines:

```r
if(!require("devtools")) install.packages("devtools")
devtools::install_github("emanuelhuber/RGPR")
library(RGPR)       # load RGPR in the current R session
```

Then inspect the console output and check that there is no error message. You can send me the output such that I can inspect it.
Thank you.

h2maa, 2021-06-11:
Dear Mr. Emanuel Huber,

Here's the screenshot of the output:
---
> if(!require("devtools")) install.packages("devtools")
Loading required package: devtools
> devtools::install_github("emanuelhuber/RGPR")
Skipping install of 'RGPR' from a github remote, the SHA1 (0bd822fc) has not changed since last install.
  Use `force = TRUE` to force installation
> library(RGPR)       # load RGPR in the current R session
Don't hesitate to contact me if you have any question:
emanuel.huber@pm.me

Attaching package: ‘RGPR’

The following object is masked from ‘package:graphics’:

    clip
---

I tried to install/uninstall RGPR many times but it did not work. I could only open .dzt and .dzx obtained from GSSI SIR-30 system but received errors when opening those obtained from GSSI SIR-4000 system.

Thank you very much.

Sincerely,
h2maa

emanuelhuber, 2021-06-11:
Strange, strange, strange.... the data you sent me, are they from the GSSI SIR-4000 system? If not, could you send them to me, such I can solve this annoying problem? Thank you.
Best,
Emanuel

h2maa, 2021-06-11:
Yes, the files that I sent you came from GSSI SIR-4000 system. If I do not delete the .dzx file, there will be an error. However, there is no problem of reading .dzt and .dzx from the SIR-30 system.

Thank you.

Sincerely,
h2maa

emanuelhuber, 2021-06-11:
OK, I will try again this evenning and I will check if I correctly updated RPGR...
Best,
Emanuel

h2maa, 2021-06-11:
Thank you, Mr. Huber! I appreciate it!!

Sincerely,
h2maa

emanuelhuber, 2021-06-11:
Hello
I pushed again some changes to github. So try to re-install RGPR. I tested and I could read your data.
Some hints:
- Are you using Rstudio? If yes, under Tools > Global option  > General, under Workspace deselect "Restore .RData into workspace at startup"
- close R
- re-open R
- try to re-install  RGPR
If it does not work, try to update the other R packages...
If it still does not work, try to re-install R.
If it is too much work for you and you only aim to export the data to .txt files, let me know, I can help you.
Best,
Emanuel

h2maa, 2021-06-11:
Dear Mr. Huber,

Thank you very much for your time and your generous offer to help. I tried all of these but they did not work. While reinstalling the RGPR, I received the below error: 

Error in i.p(...) : 
  (converted from warning) installation of package ‘raster’ had non-zero exit status

There may be an issue with my installation. I could not update the raster package from version 2.8-4 to 2.8-19. There was an error. Which version are you using now?

Sincerely,
h2maa

h2maa, 2021-06-11:
Dear Mr. Huber,

Finally, I could be able to read the .dzt file without any errors. I used RStudio in Anaconda. I do not know reasons why but it works.

Once again, I appreciate very much your time and effort to debug this issue.

Have a great weekend!

Sincerely,
h2maa

emanuelhuber, 2021-06-11:
Hello h2maa

Great! I am happy that you succeeded and kept patience.
Have an awesome weekend.
Best,

Emanuel

closed 2021-06-11 by emanuelhuber

h2maa, 2021-06-11:
Mr Huber, you have a wonderful weekend, too!

I received two errors as the following when exporting the data to .txt format:

Error in write.table(as.matrix(x), file = fPath, quote = FALSE, col.names = x@pos,  : 
  invalid 'col.names' specification

Error in xta[, 1] <- rep(pos(x), times = nrow(x)) : 
  number of items to replace is not a multiple of replacement length

When you have free time, please help me to debug these. Thanks a bunch!!

Sincerely,
h2maa

h2maa, 2021-06-11:
I tried with a different .dzx file and it works perfect. Please discard my above message.

Sincerely,
h2maa