emanuelhuber/RGPR #49: Error While Fitting Hyperbola

hhhhappyyyy, 2021-11-16:
Dear Mr. Emanuel Huber,
While fitting hyperbola using your RGPR codes, I received the below error:

> hyp <- hyperbolaFit(xy) 
Warning message:
In sqrt(1/coef1[2]) : NaNs produced

I would appreciate if you could help to debug this.
Thank you very much!

Best wish,
hhhhappyyyy

emanuelhuber, 2021-11-17:
Hello hhhhappyyyy

If you check the documentation on the function `hyperbolaFit()` you will notice that square roots are involved in the estimation of the root mean square velocity and the time of the hyperbola apex.

If somehow the points you selected cannot gives an hyperbola, there is some negative values in the square root and therefore you get the warning you mentioned.

I suggest you that you send me the coordinates of the points for which you wanted to fit an hyperbola. I can check where is the problem with the hyperbola fit.

To improve the function, I could replace the warning with a better explanation. Any additional suggestion?

Best,
Emanuel

hhhhappyyyy, 2021-11-17:
Dear Mr. Emanuel Huber,

Thank you so much for your prompt response.

I used the data （frenkeLine00.rda）which you provided. I used xy <- locator(type = "l") to obtain the coordinates of the point on the hyperbola. I clicked the position in the graph, but the correspond coordinates of the point returned were not the same as the points you selected in the tutorial. But I think the position of the points I selected give the hyperbola the same as you selected.

Here are the points you selected in the tutorial:
xy <- list( x = c( 11.8,  15.0,  17.7, 20.3, 24.4,  27.4,  30.9,  35.2),
            y = c(142.2, 119.8, 107.7, 99.5, 97.5, 105.6, 120.9, 138.1))
It can fit the hyperbola successfully.

Here are the points I selected which I used xy <- locator(type = "l"):
> xy[["x"]]
[1] -2.9604608 -1.3285596  0.2073474  1.3592777  2.8951847  4.3350976  5.2950394
> xy[["y"]]
[1] -48.40357 -56.50168 -62.28604 -64.59979 -69.22728 -63.44292 -59.97230
And I received that warning message.  

Why were the points I selected different from the points you selected? Maybe something went wrong from my side? I would appreciate if you could help to solve this issue.

Best wish,
hhhhappyyyy

hhhhappyyyy, 2021-11-17:
Hello Emanuel

Maybe I know the reason. The the points  selected by using xy <- locator(type = "l")  is different from the real coordinates of the points. I got the real coordinates of the points by using EKKO_Project 4, then I can fit the hyperbola successfully.

emanuelhuber, 2021-11-18:
OK. If you depth and horizontal axis are not correclty displayed in R, you can set the values of the axes manually.

Change the trace position values assuming a trace distance of 25 cm:
```r
pos(x) <- seq(from = 0, by = 0.25, length.out = ncol(x))
```
Change the two-way travel time or depth (here I assume a sampling interval of 0.5 ns):
```r
depth(x) <- seq(from = 0, by = 0.5, length.out = nrow(x))
```

closed 2021-11-18 by emanuelhuber