emanuelhuber/RGPR #50: Error on bits2volt function:  using argument "nbits" while it says it was supposed to be "nBytes"

felippehsk, 2022-01-12:
Hello Emanuel!

First of all, thank you so much for putting this library together. I am pretty sure that all the community appreciates it. A round of applause to you and contributors. :clap: :clap: :clap: 

It is my first time trying this library though. I have been using python for reading GSSI files, but I believe that few of the functions available here might be very useful to me. So I decided to give it a try. However, I experienced a couple of very interesting issues. I installed the library using devtools as suggested. First issue I found was a problem similar to what was reported [HERE](https://github.com/lme4/lme4/issues/407), but restarting R fixed it. I do not believe it is an issue related to your library. 

The other issue happened after installation when I was trying to open a GSSI DZT file. R produced the following error:
```
Error in bits2volt(Vmax = Vmax, nbits = x$hd$BITS) : unused argument (nbits = x$hd$BITS)
```

I found it kinda interesting so I did some digging. It seems that that instead of nbits the function should use nBytes. But what is weird the most is that in the global.R file you have in the repo, it does say 'nbits' (which is also stated on the function Description). On the other hand, when I do '?bits2volt'  in my machine it list this specific argument as 'nBytes'.  

It might be an issue with GitHub. The easiest solution I found was to I download the zip and follow [THIS](https://stackoverflow.com/a/30989367) stack overflow answer, which ends up being:

```
unzip("path to zip")
file.rename("RGPR-master", "RGPR")
shell("R CMD build RGPR")
install.packages("RGPR_0.0.7.tar.gz", repos = NULL)

```

It worked for me. This might be a one time thing. But just in case you want to look into or if anyone else encounters the same problems, I am leaving this here!

Cheers!

emanuelhuber, 2022-01-12:
Hello Felippe

Thank your very much for giving RGPR a try and for reporting the issues you got. 

The "nbits" issue is really strange. I checked the code and run some test on reading GSSI data. I cannot explain how you got this error message. Did you recently installed RGPR?

I added the link you provided (how to install RGPR from a zip file) to the companion website.

Do not hesitate to contact me if you have any question, if you are missing some functionalities or if you like to share a feedback. I am happy to help.

All the best,

Emanuel

emanuelhuber, 2022-05-12:
I close this issue since it seems that nobody faced this kind of problem.

closed 2022-05-12 by emanuelhuber