emanuelhuber/RGPR #51: Can't read .rd7 files.

roneetsingh, 2022-02-10:


emanuelhuber, 2022-02-12:
Hello roneetsingh

Thank you very much for opening this issue. Did you try to re-install RGPR and re-run your code?
If it still does not work, would you send me one of the GPR file that RGPR cannot read, in order that I find the bug?
My email: emanuel.huber@pm.me
You can count on my Swiss confidentiality. 
Best,
Emanuel

roneetsingh, 2022-02-14:
Hello Emanuel

Thank you for helping me out, To answer you, and give more details. I have done the following things:
1. Re-install RGPR and test the code for same files again.
2. Testing out the code on different .rd7 and .rd3 files.

This still does not work for the .rd7 files and some newer .rd3 files. For some older .rd3 files RGPR works like a charm. As, for the files, I have sent you an email with the files attached. I hope we can get this problem solved. Looking forward to hear back from you.
Best,
Roneet Singh

emanuelhuber, 2022-02-14:
I close this issue, as it was a problem related with the file recording and not with RGPR.

closed 2022-02-14 by emanuelhuber