emanuelhuber/RGPR #55: Error in initialize(value, ...) : object 'velocity' not found

RJbalikian, 2022-08-08:
I am trying to read in .dt1 files from Sensors and Software that worked fine just a week or so ago, and I am getting the error message in the title. I have tried several different files, in RStudio, Colab, and Visual Studio code and got the same issue.
Here is the line of code that is throwing up the error

rawGPRData <- readGPR(dsn = gprPath)

I also tried it with the sample frenke data and had the same issue.

RJbalikian, 2022-08-08:
I think the issue may be in either line 101 (uncomment it and assume data is in meters) or line 102 of readDT1, so as to include "m" as a possibility.

    if( tolower(posunit) %in% c("metre", "metres", "meter", "meters")){ 

this needs to also include "m"? At least that's what my dt1 files have. Although, I also tried editing my .hd file to say "meter" instead of "m" and that did not seem to work either.

emanuelhuber, 2022-08-08:
Thank you for opening this issue. I fixed this bug. It should work now. Please do not forget to re-install RGPR!

closed 2022-08-08 by emanuelhuber