emanuelhuber/RGPR #56: Topography and time to depth conversion

gonnavarro, 2022-09-14:
Hi @emanuelhuber ,

thanks for such needed package. 
I have a question regarding coordinates addition and time to depth conversion. 
1- It is possible to use another velocity, other than 0.1 m/ns, to convert to depth? if yes, how? if not, will this tool be available soon?
2- I have some GPR lines with GPS measurements each ~5m. I understand that I have to add this coordinates using `readFID` and then `interpPos`. I managed to do this steps but I can not plot the data with its corresponding topography, the following message appears:

> plot(DCGh3, addTopo = TRUE)
Trace vertical position set to zero!
time to depth conversion with constant velocity (0.1 m/ns)
Error in v^2 : non-numeric argument for binary operator

Any thoughts on how to fix it?

Thanks! #

emanuelhuber, 2022-09-14:
Hello @gonnavarro 

Thank you for contacting me.

To your **first question**:

You can change the velocity with the `vel()` function:

```r
vel(x) <- 0.09
```

You can also provide a velocity vector with length equal the number of samples in x (number of row, `nrow(x)`) or a list with two elements called `t` and `v` containing vector of same length defining the velocity layers. Example:

```r
vel(x) <- list( t = c(10, 95.4, 120),         # nanosecond
                v = c(0.12, 0.095, 0.07) )  # m/ns
```

Does it help you?

Normally you should be able to plot the velocity model with `plotVel()` but I just noticied, that there is a bug in this function... I will solve it.

For the **second question**, could you send me your code (and possibly a sample of your data to help me to reproduce your issue and find a solution). You can count on my confidentiality. Thank you!

emanuelhuber, 2022-09-14:
OK, I fixed the bug in `plotVel()`. Now you can plot the velocity model...

gonnavarro, 2022-09-14:
Thanks for your quick answer. 

Yes, I could use a different velocity and use it for time to depth conversion on the radargram.

Attached is the data related to the **second question** . The topography file correspond to gps points each ca 5 m, so the trace number you see in the file is an approximate of the nearest trace to each gps point (considering the real step size of the survey=0.64 m instead of the hypothetical 0.5 m). So, as I don´t have every trace gps, I have used the `readFID` and `interpPos` functions.

Thank you!

[DCGh.zip](https://github.com/emanuelhuber/RGPR/files/9570555/DCGh.zip)

emanuelhuber, 2022-09-19:
Hello @gonnavarro 

I was searching without success for a bug in my code... Now I found the problem and I am relieved! In your code try the following

```r
TOPO<- readFID('DCGh_coord.txt')
DCGh4 <- interpPos(DCGh3, TOPO)    # here I replace 'TOPO2' by 'DCGh4'
crs(DCGh4) <- "+init=epsg:4326"
plot(DCGh4, addTopo = TRUE)            # Now working
```

Do not hesitate to contact me if you have any question or if you need some specific functionalities!
Best,

Emanuel

closed 2022-09-19 by emanuelhuber

gonnavarro, 2022-09-20:
Hello again,

Yes, it has worked correctly. 
Thank you very much.

Cheers,
Gonzalo