emanuelhuber/RGPR #59: Read/import GPR data in DT1 format

AminaMS, 2023-02-19:
Hi there, what is the main steps to read my own GPR data in DT1 format?
I have using the function "readGPR()" to import GPR data into R, but I cann't clearly understand what I should do.
Can you please provides me the main steps to importing and reading my data?
Also, it is says: Error in readGPR(line1.DT1) : could not find function "readGPR" when I am typing (readGPR(line1.dt1)
Thank you.

emanuelhuber, 2023-02-20:
Hello @AminaMS 

You need to feed the function `readGPR()` with a string. A string is defined by text surrounded by double quotes:

```r
"a_string"
"a string"
not_a_string
not a string
```

That means, you should type:

```r
readGPR("line1.DT1")
```

When you type `readGPR("line1.DT1")` RGPR will check in the working directory if there is a file `line1.DT1`. To check what is your working directory, type:

```r
getwd()
```

You can change the working directory with `setwd()`. 

Alternatively, you can set the full path to your data, e.g.:

```r
readGPR("C:/Documents/GPRdata/line1.DT1")
```

Does it help you?
Best,

Emanuel

AminaMS, 2023-02-20:
Thank you for your reply, the problem is still ON!!
I have used "getwd" to know my path, it is as follows.
 getwd()
[1] "D:/A-A/A-P.hd/A-All Thesis/TGPR/GPRdata/RGPR"
Then, I used readGPR("line1.DT1"). The result is:
Error in readGPR("line1.DT1") : could not find function "readGPR"
 Then, I used readGPR("readGPR("D:/A-A/A-P.hd/A-All Thesis/TGPR/GPRdata/RGPR/line1.DT1"). The result is:
Error in readGPR("D:/A-A/A-P.hd/A-All Thesis/TGPR/GPRdata/RGPR/line1.DT1") : 
  could not find function "readGPR"")

So, would please get me the proper steps to make it well?

Thanks again ...

AminaMS, 2023-02-20:
[image]

emanuelhuber, 2023-02-21:
If you check your file, you will notice that it is "line1.dt1" (small caps for "dt1").
Therefore try:

```
readGPR("line1.dt1")
```

If you are unsure, copy the filename.

Does it solve your problem?

Best,

Emanuel

closed 2023-02-26 by emanuelhuber