emanuelhuber/RGPR #62: Plot contrast settings

bandtank, 2023-04-06:
Is there a way to change the contrast of the data points? It is difficult to see some of the lines in the plot using the RGPR library, but I don't have the same issue with commercial software. I'm not sure if there are other factors at play or what I could be doing differently. Here is an example of the RGPR output:

[image]

emanuelhuber, 2023-04-06:
Thank you very much for raising this issue.
You can clip the signal amplitude. For example:

`plot(x, clip = 0.01)`

That is a linear transformation. Let me know if it solves your problem.
If not, I will find a solution.

Best,

Emanuel

bandtank, 2023-04-06:
That worked. Thanks.

For anyone else who wants to see the documentation of the clip functionality, go [here](https://github.com/emanuelhuber/RGPR/blob/363beed8cf379535bd84bad08870ff4cdce56f5f/R/plotly.R#L17).

closed 2023-04-06 by bandtank

emanuelhuber, 2023-04-06:
To check the documentation about a function, enter in the r console

```r
?functionName
```

For example

```r
?plot
```

In the documentation the argument `clip` is described.