emanuelhuber/RGPR #63: Changing axes from time to distance

bandtank, 2023-04-06:
Is there an official way to change the axes from time to distance? The input format is GSSI. Each of my dzt files has an accompanying dzx file, which contains information like this:

```
<?xml version="1.0" encoding="UTF-8"?>
<DZX xmlns="www.geophysical.com/DZX/1.02">
<GlobalProperties>
   <verticalUnit>in</verticalUnit>
   <horizontalUnit>in</horizontalUnit>
   <dielectric>6.500000</dielectric>
</GlobalProperties>
<DataCollection>
   <system>SS_MINI</system>
   <antennaType>HiRes</antennaType>
   <WheelSize>NORMAL</WheelSize>
   <softwareVersion>1 4 4 0</softwareVersion>
   <collectionMode>2D</collectionMode>
   <sLastUpdateDate>MAR_7_2023</sLastUpdateDate>
   <sLastUpdateTime>10:42:12</sLastUpdateTime>
   <BaseGain>13.500000</BaseGain>
   <BaseGainAdj>0.100000</BaseGainAdj>
   <TrOffset>36757</TrOffset>
   <dielectricMethod>0</dielectricMethod>
   <Ctable>3</Ctable>
   <Xtable>2</Xtable>
</DataCollection>
</DZX>
```

It says the vertical and horizontal units are `in`, but I think the time units in the default plot are correct. I can't find an API in RGPR that helps with unit conversion.

emanuelhuber, 2023-04-06:
Thank you for your question.

RGPR automatically converts the inches into meters. Unit conversion is not yet fully implemented in the current RGPR version (to switch from meters to inches or from nanoseconds to miliseconds). This will be fully supported in the upcoming version.

Now it is not clear for me, what you exactly want to do. Do you want to convert the time unit (vertical axis) into depth? Then use the function 'migrate()'. You can specify a wave velocity with 

`vel(x) <- 0.09`

Do not hesitate to contact me if you need some specific functionalities.

Best,

Emanuel

bandtank, 2023-04-06:
Thanks for the information. That helps.

The default unit for the horizontal axis is nanoseconds, but the units appear to be meters, which matches your explanation. Look at the following data:

```
> print(files[1])
[1] "test-data//File____112.DZT"
> gprData <- readGPR(dsn = files[1])
Antenna separation set to 0 m. Set it with 'antsep(x) <- ... '
> gprData
*** Class GPR ***
 name        = File____112
 filepath    = test-data//File____112.DZT
 41 fiducial(s)
 description = 
 survey date = 2023-03-07
 reflection, 388 MHz, Window length = 4.96078431372549 ns, dz = 0.0196078431372549 ns
 4874 traces, 6.18871019264174 m
```

Here is the plot:
[image]

The horizontal axis looks like it ranges from 0 to about 6.2, which matches `6.18871019264174 m` as shown in the data. Is this what you expect? If so, should I always set the label to meters?

Regarding the vertical axis, yes, I want to convert the time unit to depth. I see the explanation in the documentation now that I know which word to use (`migrate`). I'll come back if I can't figure it out.

emanuelhuber, 2023-04-06:
Thank you for the information. The x-label with 'ns' is indeed very strange.
Which function did you used to plot the data? `plot()` or `plotFast()`?

Can you show me the output of `x@posunit`, where `x` is your GPR object.

bandtank, 2023-04-06:
```
library(RGPR)

amplitude = 0.01
palette = "grey3"
dataPath <- "test-data/"

files <- list.files(
  dataPath,
  pattern = "^.*\\.dzt$",
  ignore.case = TRUE,
  full.names = TRUE
)

for(file in files) {
  gprData <- readGPR(dsn = file)
  fileName <- tools::file_path_sans_ext(basename(file))
  plotFast(
    gprData,
    col = palGPR(palette),
    #clip = amplitude
  )
}
```
Produces:
[image]

Changing to `plot` produces:
[image]

A few observations:
- `plotFast` does not accept `clip`, so I had to comment that out
- `plot` shows `meters (m)` as the x-axis label with no changes other than including `clip`
- The plots look pretty different in terms of styling and contrast.

emanuelhuber, 2023-04-06:
Thank you!

* You can apply `clip()` to the data before ploting:
    ```r
    plotFast(clip(gprData, 0.01))
    ```
* I fixed the bug in  `plotFast()`. You have to re-install RGPR.
* You can set the x-axis and y-axis labels manually with:
    ```r
    plot(gprData, xlab= "my xlabel", ylab = "my ylabel")
    # or
    plotFast(gprData, xlab= "my xlabel", ylab = "my ylabel")
    ```
* The yellow arrows are fiducial markers (markers that were set during survey). To remove them, use:
    ```r
    plot(gprData, addFid = FALSE)
    ```
Do not hesitate to contact me if you have any question!

bandtank, 2023-04-06:
This is great stuff. Thank you for all the help. The library is extremely useful.

closed 2023-04-06 by bandtank