emanuelhuber/RGPR #67: Can't read .dt1 files

moodyj, 2023-08-16:
I'm having issues reading some Sensors and Software dt1 files. They were extracted from the newer GPZ format (just a renamed zip file). The profiles look fine in pulseEKKO so the data is ok. The HD file looks normal, as far as I can tell. RGPR, GPRpy and RADAN (version 6 so quite old) will open the file with no problem but the plots are nonsense. I've attached an example line for investigation.
[liney47.zip](https://github.com/emanuelhuber/RGPR/files/12362999/liney47.zip)

moodyj, 2023-08-17:
I got it figured out. This version of the dt1 file uses a 32bit float to store data vs 16bit integer for the other version. The critical line to change is 322 of readDT1.R:
`dataDT1[,i] <- readBinary(dsn, what = "integer", n = npt, size = 2)`
which will work with
`dataDT1[,i] <- readBinary(dsn, what = "numeric", n = npt, size = 4)`
There are also changes required to how the function estimates the number of points and traces but I don't have the R chops to make it work with both versions of the dt1 files.

emanuelhuber, 2023-08-21:
Hello @moodyj 
Thank you for your feedback! I will add a "if-else-loop" to handle the 16 bits / 32 bits (maybe this evening).

All the best,

Emanuel

emanuelhuber, 2023-08-21:
Hello @moodyj 

I just updated RGPR, normally you should be now able to read 32 bits. Let me know if it works.
Best,

Emanuel

closed 2023-09-24 by emanuelhuber