emanuelhuber/RGPR #68: plot `SGY` error

fengziren, 2023-09-25:
Today I used the same approach and found that drawing the image reported an error

input
```
library(RGPR)
search()
x <- readGPR(dsn = "~/Recordings/3.sgy")

print(x)
plot(x)
```
output
```
Don't hesitate to contact me if you have any question:
emanuel.huber@pm.me

Attaching package: ‘RGPR’

The following object is masked from ‘package:graphics’:

    clip

Antenna frequency set to 0 MHz. Set it with 'antfreq(x) <- ... '
Antenna separation set to 0 m. Set it with 'antsep(x) <- ... '
*** Class GPR ***
 name        = 3
 filepath    = ~/Recordings/3.sgy
 description = 
 survey date = 2023-01-25
 reflection, 0 MHz, Window length = 49.64 ns, dz = 0.085 ns
 4559 traces, 4558 m
 ****************
Error in base::rowSums(x, na.rm = na.rm, dims = dims, ...) : 
  'x' must be an array of at least two dimensions
```
However, when I use `plotFast` it draws the image successfully!

I was able to use `plot` plotting normally on the day 2023-9-24, I didn't do anything during this period, I don't know what happened, please tell me what to do, thanks!
[3.zip](https://github.com/emanuelhuber/RGPR/files/12714197/3.zip)

emanuelhuber, 2023-09-25:
Thank you @fengziren for raising this issue! I will allocate some times tomorrow to check that. Thank you for your patience.
Best,
Emanuel

fengziren, 2023-09-26:
> Thank you @fengziren for raising this issue! I will allocate some times tomorrow to check that. Thank you for your patience.感谢您提出这个问题！明天我会分配一些时间来检查这一点。感谢您的耐心等待。 Best, 最好的， Emanuel  伊曼纽尔

Dear Mr. Emanuel Huber

Thank you very much

Bast wish,
fengziren

emanuelhuber, 2023-09-26:
Hello @fengziren 

I updated the function to read SEG-Y on Sunday to make it able to extract the coordinates from the SEG-Y files. I forgot to handle the case when all the coordinates are (0, 0, 0) which caused the error in the plot function. It is now solved.

Note that you need to re-install RGPR before running your code:

```
devtools::install_github("emanuelhuber/RGPR")
library(RGPR)
```

Do not hesitate to contact me if you have any question. Happy to help!

Best,

Emanuel