emanuelhuber/RGPR #69: Request for Support: Adding Support for Easy Radar USA's .gpr Format

fengziren, 2023-11-06:
Dear Mr. Emanuel,

I hope this message finds you well. I am writing to request your assistance in implementing support for the .gpr format radar files generated by Easy Radar USA, a company specializing in radar solutions.

Our team is currently working on a project that involves analyzing radar data, and we have recently started using Easy Radar USA's equipment. However, we have encountered a challenge as our existing tools and software do not have built-in support for the .gpr format.

Best regards,

fengzireng
[2023214173934.zip](https://github.com/emanuelhuber/RGPR/files/13262270/2023214173934.zip)

emanuelhuber, 2023-11-12:
Hello @fengziren 

Thank you for contacting me. I would be more than happy to add the ".gpr" data format. For that I need to get at least an idea of the binary data format. A binary file is a kind of a black box, a series of 0 and 1. 
Could you try to find out, what is the binary data format of the .gpr format? Could you ask the manufacturer? I need to know which byte carries which information.

Thank you!

Best,

Emanuel

fengziren, 2023-11-13:
> Hello @fengziren
>
> Thank you for contacting me. I would be more than happy to add the ".gpr" data format. For that I need to get at least an idea of the binary data format. A binary file is a kind of [...]

Dear Mr. Emanuel

Thank you very much for your support, I have tried to seek help from the manufacturer but have not received a positive response

Best regards,

fengzireng

closed 2023-11-13 by fengziren

emanuelhuber, 2023-11-14:
Hello @fengziren 

I just sent an email to the manufacturer. Let see if I will get an answer....
You also could check if the .gpr file format is not in fact a .sgy file (change the extension and try to read it with RGPR).

I will try to keep an eye open if somewhere on the web the binary description of the .gpr format appears.

Best,
Emanuel

fengziren, 2023-11-15:
Thanks @emanuelhuber 

I've tried changing the extension and trying to read it with RGPR with no success!

Thanks again for your support and I will sync with you as soon as I find something new!

Best regards,

fengzireng

emanuelhuber, 2023-11-24:
I got the following answer from Easy Radar USA:

> We are just ambot done covering our data to be compatible with standard segy format. I will send you files soon to make sure they work with your program

So let see.

Best,

Emanuel

fengziren, 2023-11-27:
Thanks @emanuelhuber

Thank you once again for your exceptional help. Your contributions have been instrumental in resolving the challenges I faced, and I am grateful for your generosity and expertise.

Best regards,

fengzireng