emanuelhuber/RGPR #70: Is the dielectric value in dzx files used?

bandtank, 2023-11-21:
I see how the dzx/dzt files are parsed, especially the global properties, at [this](https://github.com/emanuelhuber/RGPR/blob/bb7e8b4abd4b35cdcdcea0893cb833296b1ea55b/R/readDZT.R#L432) line. It doesn't look like the dielectric value is used in RGPR, but I figured I should ask just in case I am missing it. If it is not used, is there a reason why? Perhaps it is assumed the user will do something with the dielectric value before or after the data has been processed by RGPR. Any help would be appreciated.

emanuelhuber, 2023-11-24:
Hello @bandtank 

Thank you for your question. Yes, currently, the dielectric value in the dzx file is not parsed and not used in RGPR. Let me know if I am wrong: the relative dielectric permittivity `u` could be used to provide an estimate of the electromagnetic propagation velocity 

`v = c0/sqrt(u)`

So either the user compute by itself the EM propapation velocity or I update RGPR so that it reads the relative dielectric permittivity and estimates the velocity. What do you mean?

Best,

Emanuel

bandtank, 2023-11-25:
Thanks for the response. You're right - the user could calculate `v` using the given equation. For anyone else who wants to learn more, [here](https://www.guidelinegeo.com/help-articles/what-is-the-dielectric-constant-and-why-it-matters-to-gpr/) is a good explanation.

I don't think this is a critical issue. With that said, it would be cool if RGPR could compute velocity using the dielectric permittivity from the dzx file. If I have time over the next few weeks, I will submit a PR with the updates.

emanuelhuber, 2023-11-26:
Thank you for your feedbakc @bandtank !
I will update RGPR such that the dielectric value found in the raw data is used to estimate the EM-wave velocity. Please do not close this issue... so I have a reminder! ;-)
Best,
Emanuel