emanuelhuber/RGPR #78: incorporate topography data

arnab-pal, 2024-06-26:
I want to incorporate the topography data in the GPR data. Please help how to incorporate the data.

emanuelhuber, 2024-06-28:
Hello @arnab-pal 

Thank you for your question. Here there is a tutorial on how to add coordinates (x, y, z) to GPR data (that is how to add topography):
https://emanuelhuber.github.io/RGPR/04_RGPR_tutorial_GPR-data-survey/

After some processing (optional), you can display your data with the topography with the `plot()` function (set the argument `addTopo = TRUE`). You can also plot your data in 3D with the `plot3DRGL()` function.

You can transform you data from time to depth with the `migrate()` function. 
If you set the attribute 

- `type = "static"`  you get a trace-by-trace time-to-depth conversion that shifts the traces vertically according to their elevation.
- `type = "kirchhoff"` a topographic Kirchhoff migration is applied
-
Check the documentation of the function `migrate()` with.
```
?migrate
```

Does it help you?

Best,

Emanuel

arnab-pal, 2024-06-28:
Hello Emanuel,
Thanks for the information. I want to clarify that I already processed the GPR data by following the above steps. The GPR data is showing depth information from 0 to 40 meter. I have separate file of topography with distance and elevation information. The elevation is for the top where the GPR survey was performed. How I will incorporate that data into the GPR file using RGPR.

Thanking you

Arnab Kumar Pal

Research Scholar,

Earth System Science and Engineering<https://www.iitg.ac.in/civil/Earth_Sciences_Lab/index.html>

Department of Civil Engineering<http://www.iitg.ac.in/civil/site/>

IIT Guwahati<http://www.iitg.ac.in/>, Assam 781039, India

Phone: 0361-258-3343/3344 (Lab)

Mobile: 8900602360

Web:

https://site.arnab_pal<https://sites.google.com/view/arnab-kumar-pal>

https://www.researchgate.net/profile/Arnab-Pal-14

https://emanuelhuber.github.io/RGPR/04_RGPR_tutorial_GPR-data-survey/

emanuelhuber, 2024-06-28:
Hello Arnab

So, you have GPR data, `x`, that is processed but has no coordinates (= no topography data). Is that correct?

You also have a file containing the distance and elevation information for **every traces**, is that correct?

My suggestion:
```
# DZ = two column vector
DZ <- read.table("your_file.txt", sep = "\t", header = TRUE)

# create a coordinate matrix (GPR data is along y-direction)
xyz <- cbind(0, as.matrix(DZ))

# set coordinates (works only if ncol(x) == nrow(xyz)
coord(x) <- xyz

# plot with topography
plot(x, addTopo = TRUE)
```

arnab-pal, 2024-06-28:
Dear Emanuel,
Yes, I have modified the code according to you. Is there any requirement that GPR file should be in depth domain, or it will automatically convert it to depth from time axis. As after running the code in the R, it is showing "time to depth conversion with constant velocity (0.100000001716614 m/ns)". I have used the RAW GPR file with no time to depth conversion.

#read the GPR file with the suitable extension
gpr_data <- readGPR("F:/arnab/PROJECT0ARN__001.DZT")

# DZ = two column vector
DZ <- read.table("F:/arnab/Elevation1.txt", sep = "\t", header = TRUE)

# create a coordinate matrix (GPR data is along y-direction)
xyz <- cbind(0, as.matrix(DZ))

# set coordinates (works only if ncol(x) == nrow(xyz)
coord(gpr_data) <- xyz

# plot with topography
plot(gpr_data, addTopo = TRUE)

Thanking you

Arnab Kumar Pal

Research Scholar,

Earth System Science and Engineering<https://www.iitg.ac.in/civil/Earth_Sciences_Lab/index.html>

Department of Civil Engineering<http://www.iitg.ac.in/civil/site/>

IIT Guwahati<http://www.iitg.ac.in/>, Assam 781039, India

Phone: 0361-258-3343/3344 (Lab)

Mobile: 8900602360

Web:

https://site.arnab_pal<https://sites.google.com/view/arnab-kumar-pal>

https://www.researchgate.net/profile/Arnab-Pal-14

emanuelhuber, 2024-06-28:
Hello Arnab

There is no requirement that the GPR data has to be in time domain. If the GPR data is in time or depth domain is independent of the coordinates. Normally, when you read raw GPR data, they are in time domain. When you apply `migrate()`, RGPR switch to the depth domain (i.e. adapt the units).

Best,

Emanuel

arnab-pal, 2024-06-28:
Dear Emanuel,
Thanks for the information. After performing the topographic correction, the GPR section is showing 0 on the top and 60 meter on the bottom as elevation. How can I rectify that.

Thanking you

Arnab Kumar Pal

Research Scholar,

Earth System Science and Engineering<https://www.iitg.ac.in/civil/Earth_Sciences_Lab/index.html>

Department of Civil Engineering<http://www.iitg.ac.in/civil/site/>

IIT Guwahati<http://www.iitg.ac.in/>, Assam 781039, India

Phone: 0361-258-3343/3344 (Lab)

Mobile: 8900602360

Web:

https://site.arnab_pal<https://sites.google.com/view/arnab-kumar-pal>

https://www.researchgate.net/profile/Arnab-Pal-14

emanuelhuber, 2024-06-28:
Hello Arnab

That's the depth domain. To display your data according to the elevation (and not depth) try:
```
plot(x_migrated, elev = TRUE)
```
Best,
Emanuel

arnab-pal, 2024-06-29:
Hello Emanuel,

Here X_migrated means I have to perform migration operation on the data to finally visualise the data according to the elevation of the section.

Thanking you

Arnab Kumar Pal

emanuelhuber, 2024-06-29:
Hello Arnab

You do not need to migrate the data. If you only want to plot the data (and do not need a Kirchhoff migration), you can simply run

```
plot(x, elev = TRUE)
```
Where `x` is your non-migrated data with coordinates (x, y, z). When you set `elev = TRUE` each trace is converted to depth domain.

Best,

Emanuel

arnab-pal, 2024-06-29:
Hello Emanuel,
Thanks for the help. I have another quarry whether the final GPR file can be exported in SEG-Y format in R environment.

Thanking you

Arnab Kumar Pal

emanuelhuber, 2024-06-29:
You can do that with:
```
writeGPR(x, fpath = "DAT_0448_test2.sgy")
```
where `x` is your GPR data.

Best,

Emanuel

arnab-pal, 2024-07-02:
Hello Emanuel,
How can I crop the data in Y axis and save it separately for my purpose?

Thanking you

Arnab Kumar Pal

Research Scholar,

Earth System Science and Engineering<https://www.iitg.ac.in/civil/Earth_Sciences_Lab/index.html>

Department of Civil Engineering<http://www.iitg.ac.in/civil/site/>

IIT Guwahati<http://www.iitg.ac.in/>, Assam 781039, India

Phone: 0361-258-3343/3344 (Lab)

Mobile: 8900602360

Web:

https://site.arnab_pal<https://sites.google.com/view/arnab-kumar-pal>

https://www.researchgate.net/profile/Arnab-Pal-14

emanuelhuber, 2024-07-02:
You can use the `crop()` function:

```
xcrop <- crop(x, xlim = NULL, ylim = c(0, 10))
```

Best,
Emanuel

arnab-pal, 2024-07-02:
Hello Emanuel.

I have followed the same. But after performing the correction, while plotting the modified data it is showing the error.

 plot(gpr_data001_1)
Error in if (dots$ylim[1] < dots$ylim[2]) dots$ylim <- rev(dots$ylim) :
  missing value where TRUE/FALSE needed

Thanking you

Arnab Kumar Pal

arnab-pal, 2024-07-03:
Hello Emanuel.

I have .dzg file with GPR latitude, longitude and elevation information. How to read the file and incorporate it to the existing gpr file for topography correction in R environment. I have attached screenshot of the .dzg file opened in text file format.

[cid:14103cc9-28ec-4b49-847f-cac08d9d976a]

Thanking you

Arnab Kumar Pal

emanuelhuber, 2024-07-04:
Hello Arnab

If your .dzg file has the same name as your GPR data (for example "data01.dzt" and "data01.dzg") and if you set `interp_pos=TRUE` in `readGPR()`, then RGPR reads both file and add coordinates to the traces. 

```
x <- readGPR("data01.dzt", interp_pos = TRUE)
x
plot(coord(x))
```
Let me know if it works.

Alternatively you can do that:
```
gps <-  readDZG("data01.dzg", toUTM = TRUE)
x <- interpPos(x, gps$mrk)
```

Best,

Emanuel

closed 2024-08-04 by arnab-pal