emanuelhuber/RGPR #84: Add a license file

HamishB, 2024-09-16:
Hi, would you be open to adding a license file into the code repository?

It's not clear to me what it is, and I'd like to know that before I spend time contributing anything.
(I'd be happy for it to be the same license as R itself, but if it's not explicitly stated it is unknown)

Many thanks.

emanuelhuber, 2024-09-28:
Helloe @HamishB 

Please apologize my delayed answer, I was quite busy last weeks.
RPGR is under the license GPL (>= 2), see the description file. In R-package, the licence is in the DESCRIPTION file.

Is that good for you? 

I would be very happy for any contribution!

Best

Emanuel

HamishB, 2024-11-02:
Perfect, thanks. I hadn't looked in the DESCRIPTION file.

regards,
Hamish

closed 2024-11-03 by emanuelhuber